intel/node-realsense #21: [SLAM]Should get reject promise when invoke  loadOccupancyMap()/loadRelocalizationMap() without parameter

zhaoming0, 2017-02-23:
According spec:
```
Promise<void> loadOccupancyMap(String mapFileName);
Promise<void> loadRelocalizationMap(String mapFileName);
```

Method should have a mandatory parameter input, when missing it, method should return reject promise

Test Code:
```
let argv = '/tmp/loadOccupancyMap.map';      
     slamAddon.createInstance().then((Instance) => {
        slamInstance = Instance;
        return slamInstance.start();
      }).then(() => {
        return slamInstance.saveOccupancyMap(argv);
      }).then(() => {
        return slamInstance.loadOccupancyMap();
      }).then((Data) => {
        reject('Promise should reject, but got resolved');
      }).catch((err) => {
        resolve();
      }); 
```
Result:
`Error: the string "Promise should reject, but got resolved" was thrown, throw an Error :)`

DonnaWuDongxia, 2017-03-13:
As #153 merged, please double check if this bug still exist. @zhaoming0 Thanks!

haoyunfeix, 2017-03-27:
@lisa0314 and @DonnaWuDongxia , this issue could reproduce on 0.10.0 pre-release.
Case located at https://github.com/01org/node-realsense/blob/master/src/slam/test/Instance/test-loadOccupancyMap.js#L52, please remove the ".skip" when you try it.

DonnaWuDongxia, 2017-03-28:
@haoyunfeix we changed the interface, https://github.com/01org/node-realsense/blob/master/src/idl/slam/slam.widl#L129, only two interfaces(getOccupancyMap[Update] ) have optional arguments, so the test cases should be modified accordingly.

haoyunfeix, 2017-03-28:
@DonnaWuDongxia - This is a Negative test case, we specially design it to see what will happen when call a method without "Mandatory" argument. We expected to receive "reject" promise in this pattern.

DonnaWuDongxia, 2017-03-30:
Verified with @kenny-y that we need to modify the WILD-NAN tool to fix this bug, so move it out of this milestone.