intel/node-realsense #22: [SLAM] saveRelocalizationMap() should return string but got a number

zhaoming0, 2017-02-23:
According spec:
`Promise<String> saveRelocalizationMap(optional String mapFileName);`

Should got string but return a number

Test Code:
```
     let options = '/tmp/saveRelocalizationMap.map';
        slamAddon.createInstance().then((Instance) => {
          slamInstance = Instance;
          return slamInstance.start();
        }).then(() => {
            if (options == null ) { 
              return slamInstance.saveRelocalizationMap();
            }else{
              return slamInstance.saveRelocalizationMap(options);
            }   
        }).then((Data) => {
          assert.equal(typeof(Data), 'string');
          resolve();
        }).catch((err) => {
          reject(err);
        }); 
```

Result:

`AssertionError: 'number' == 'string'`

lisa0314, 2017-02-24:
@zhaoming0, thanks for your test. But we have not implemented `saveRelocalizationMap` by now. I think we can retain this issue until we implemented the API.

DonnaWuDongxia, 2017-03-13:
As #153 merged, please double check if this bug still exist. @zhaoming0 Thanks!

closed 2017-03-14 by haoyunfeix