intel/node-realsense #5: Enable Travis CI

halton, 2017-02-23:
@baleboy could you help to setup on Travis CI since you have admin permission.

1. Visit https://travis-ci.org
2. Toggle 01org/node-realsense project.
3. Click the settings (right side of the toggle button in step 3), toggle "Build only if .travis.yml is present".

Then all should works. For more information, please refer to https://docs.travis-ci.com/user/travis-pro/

baleboy, 2017-02-27:
Apparently I don't have admin rights on the repo, I'll have to request them.

baleboy, 2017-02-27:
This is done now, please check if it works

halton, 2017-02-27:
@baleboy thanks a lot!

halton, 2017-02-27:
[image]

@baleboy it shows me it still disabled.

baleboy, 2017-02-27:
uhm, yes, I was able to configure the "build only if .yml is present" but not the master switch. Let me try again.

baleboy, 2017-02-27:
Travis and Github were out of sync, now it should *really* be OK

halton, 2017-02-27:
Now it is working, will close this bug. Thanks a lot for your supporting. @baleboy

closed 2017-02-27 by halton