intel/node-realsense #79: [SLAM] Adopt the naming change request of last WIDL review

haoyunfeix, 2017-02-24:
Fork this from #76 ...

PT is the forerunner to adopt the naming change, now there is no `interface Instance{}` in PT add-on but a more meaningful `interface PersonTracker {}`

We should set a milestone to make OR/SLAM to adopt the similar naming change as well.

DonnaWuDongxia, 2017-02-27:
interface name changing:
`createInstance  --> getSlam[Tracker] `

Class name changing:
`Instance  --> Slam[Tracker] ?`

package name (npm module name) changing:
`slam --> realsense-slam`

@halton @kenny-y  WDYT?

halton, 2017-02-27:
> package name (npm module name) changing:
slam --> realsense-slam

As today's quick sync with @kenny-y @huningxin, we agree to use npm module name as: `node-person`,  `node-object` and `node-slam`