intel/node-realsense #113: [Sample][SLAM] Missing the 'input' and 'tracking' status in browser

haoyunfeix, 2017-02-26:
Missing the `input` and `tracking` status in browser when running sample of [slam_tutorial_1_web] on Joule Board with 0.8.0 release, this issue cannot reproduce on native.
native:
[image]
0.8.0 release:
[image]

DonnaWuDongxia, 2017-03-13:
Is this means the camera pose is floating while actually the camera hasn't been moved?

haoyunfeix, 2017-03-15:
Oh, I mean the status bars (on top of Fisheye snapshot) didn't show data.
Input: fish, depth, gyr., acc.
Output: depth

DonnaWuDongxia, 2017-03-30:
It's in expectation that we have no Input: fish, depth, gyr., acc.
The depth Output should be ready in current master, would you please verify it? @haoyunfeix

haoyunfeix, 2017-03-30:
Verified with master branch, close this issue.

closed 2017-03-30 by haoyunfeix