intel/node-realsense #119: Requirement of a type of user created async task that fufiled by async callback of MW

haoyunfeix, 2017-02-26:
There is a user scenario of PT, it is caused by the asychronous APIs of MW, which may need to enhance our async task framework.

User could need the MW recognize all persons in in the FOV of the camera, which operation is time consuming. Thus the MW provided an asynchronous API, which can trigger this operation, and the result is carried in a callback method which may be called in different threads other than the worker thread. And when the result is got, there may already passed several frame cycles.

This task is user created task, with promise as return value. And when this task is executing, it calls MW's method, and then task maybe put to a state like pending. And the worker thread continues to process other tasks. Some time later, a callback was called with the result carried, the task shall be marked as successful or failed, and the task may need to be put to the worker cycle to be processed again to resolve/reject the returned promise. This may involve state changed to be ready.

the flow can be described as:
User created task => execute in worker thread (call MW's method) => state changed to pending => ...time passes... => result call back from MW => execute again in worker thread to consume the result => resolve/reject promise in main thread.

tingshao, 2017-02-27:
@kenny-y I had a initial idea for this feature, can you provide your comments? If the solution is ok, I will implement it. Thanks.

the design in my thought include:
1.  Add a new AsyncTask type to encapsulate necessary code. This kind of task is set to pending after execution, and state may be changed after MW done some work asynchronously. The state change behaviour should be done in worker thread.
2. Add a new uv_async_t with worker thread, and the associated callback would be used to do some work for the pending tasks, for example to consume the result generated by the MW, and update the task's state. I did consider to reuse the old uc_ascyn_t, but in that way, we need to alter the existing runner loop logic of work thread (the OnAsyncNotification): to not only process Ready tasks, but also Pending tasks.
3.  add a public or at least friend API to AsyncTaskRunner for use by the new AsyncTask, like SendAsyncNotification() to enable other threads (MW's inner thread) to trigger async notification on the worker thread. This is used to trigger the callback created in step 1 above.

In the above idea, I though there may be future need of a task that is not a one shot work, but may continues to do something, a little and a little which involve multiple loops. And the MW may always invole in a asynchronous way. For example download and process a large data section. The user may trigger a task, the task is pending in worker thread, and the mw repeatedly callback to update with more data, the worker thread consume the data, and still keeps to be pending. a long time later, after everything is done, the task is set to Successful. Currently we seems have no such request, but the above design seems easy to be extended for that purpose.

kenny-y, 2017-03-06:
Suggested change (noted down from the quick meeting)

```
class AsyncTask {
  virtual void UpdateTaskState() {};
}

class TwoStepPromiseTask {
 public:
}

AsyncTaskRunnerD::ActivateWorkerThread() {
  abort_destroy_threads_ = true;
  TouchThreads();

  worker_thread_->SendAsyncNotification("{ACTIVATE_WORKER_THREAD}");
}

AsyncTaskRunner::ActivateWorkerThread() {
  d_->ActivateWorkerThread();
}

```

tingshao, 2017-03-08:
I rechecked the native SDK, and found we may over estimated the needs of the TwoStepPromiseTask: the recognition only happens in one frame, multi-frame is not supported yet. And in their sample, the registration work is also synchronous, so, I will keep to use the old normal promise task, and this design can be helpful for future possible needs. Thanks kenny's help.

tingshao, 2017-03-15:
Close this as it's obsoleted.

closed 2017-03-15 by tingshao