intel/node-realsense #132: [PT] 14 test cases filed on Travis

halton, 2017-02-27:
https://travis-ci.org/01org/node-realsense/builds/205734871

```
55 passing (1m)
  1 pending
  14 failing
  1) Person Tracking Test Suite - State - Full create - start - reset:
     Error: Promise rejected with no or falsy reason
  
  2) Person Tracking Test Suite - State - Full create - start - resume:
     Error: the string "Already started" was thrown, throw an Error :)
  
  3) Person Tracking Test Suite - State - Full create - start - pause - stop:
     Error: the string "Already started" was thrown, throw an Error :)
  
  4) Person Tracking Test Suite - State - Full create - start - start:
     Error: the string "Already started" was thrown, throw an Error :)
  
  5) Person Tracking Test Suite - State - Full create - start - pause - resume:
     Error: the string "Already started" was thrown, throw an Error :)
  
  6) Person Tracking Test Suite - State - Full create - start - stop- start:
     Error: the string "Already started" was thrown, throw an Error :)
  
  7) Person Tracking Test Suite - State - Full create - start - pause - start:
     Error: the string "Already started" was thrown, throw an Error :)
  
  8) Person Tracking Test Suite - State - Full create - start - pause - pause:
     Error: the string "Already started" was thrown, throw an Error :)
  
  9) Person Tracking Test Suite - States Start => Stop:
     Error: the string "Already started" was thrown, throw an Error :)
  
  10) Person Tracking Test Suite - States start => pause => resume => stop:
     Error: the string "Already started" was thrown, throw an Error :)
  
  11) Person Tracking Test Suite - States start => reset => start => stop:
     Error: the string "Already started" was thrown, throw an Error :)
  
  12) Person Tracking Test Suite - tracking Start => getPersonInfo rejected:
     Error: the string "Already started" was thrown, throw an Error :)
  
  13) Person Tracking Test Suite - tracking Not crash when tracking is disabled:
     Error: the string "Exception happened" was thrown, throw an Error :)
  
  14) Person Tracking Test Suite - tracking detectionMode coverage test:
     Error: the string "Failed to cover all detection mode,Already started" was thrown, throw an Error :)
```

kenny-y, 2017-03-07:
Is this issue still valid?

halton, 2017-03-07:
Check the latest travis status, yes, but one less failure.
```
  63 passing (51s)
  5 pending
  13 failing
```