intel/node-realsense #145: Make any number of the 3 add-on(s) work at the same time (multi-model support)

kenny-y, 2017-03-07:
Could be:
- OR + PT
- OR + SLAM
- OR + PT + SLAM
- PT + SLAM

librealsense add-on: TBD

kenny-y, 2017-03-07:
Started working on this today... aims to achieve a quick prototype within days...

kenny-y, 2017-03-09:
Process level object singleton sharing across multiple builds -  done
New delegate class of camera control - able to run on single add-on.

kenny-y, 2017-03-10:
[image]

tingshao, 2017-03-13:
Excited to know that!

halton, 2017-03-13:
Super

kenny-y, 2017-03-14:
Hit more core dump than before, after enabled PT + OR work together. The source line is: 

https://github.com/BVLC/caffe/blob/master/src/caffe/common.cpp#L58

No clue at present.

kenny-y, 2017-03-14:
SLAM module is now able to build and run with other modules, but with an error state:

The following condition is always true in frame_callback @DonnaWuDongxia 

```
if (rs::timestamp_domain::microcontroller != timestampDomain) {
```

kenny-y, 2017-03-15:
Finally made 3 addons work together, with arbitrary startup sequence -- formerly there was a limitation that SLAM must be started up as the 1st addon.

Please be noted that I believe the orthogonal solution is the most elegant one...

kenny-y, 2017-03-16:
Introduced a new helper class to manage the requirement among different add-ons, will PR tomorrow to make sure that I'm lucent...

kenny-y, 2017-03-16:
`mocha` failed on individual add-on (running in `node` is OK), need to figure it out... PR is pending...

commit 32dba7e7 (2017-03-21): Multi model support (#163)

* Add support to enable 3 addons run in arbitrary order

Closes #145
closed 2017-03-21 by kenny-y