intel/node-realsense #159: [Open] What's the version number for next mileston March 30th.

DonnaWuDongxia, 2017-03-14:
As we decided to make next milestone on March 30th. Now our version number has bumped to 0.9.9 in the commit log. So what's the version number shall we use for next milestone, 1.0.0?
What's your opinions, @halton @kenny-y ? Can not find Ningxin in the group.

halton, 2017-03-14:
1.0.0 seems to be more stable release, are we sure stable stage? If no, let's use 0.10.0

minggangw, 2017-03-15:
Will the node-librealsense be added to this release at the end of Mar?

halton, 2017-03-15:
> Will the node-librealsense be added to this release at the end of Mar?

Depend on you

minggangw, 2017-03-15:
I will contact the QA team to make sure if we can finish the first round test for the existing functions before this milestone.

DonnaWuDongxia, 2017-03-15:
So, I will create a milestone on 0.10.0

DonnaWuDongxia, 2017-03-15:
@kenny-y @tingshao @lisa0314 @minggangw [0.10.0 milestone](https://github.com/01org/node-realsense/milestone/2) has been created, please add scoped issues to it according your judgements.

minggangw, 2017-03-17:
I just negotiated with @haoyunfeix about testing work of node-librealsense. We decided to design and run the test cases for the existing APIs first, and if the testing work can be finished before this milestone (0.10.0), we can release it. After this release, other functions and TCs will be added step by step.

halton, 2017-03-17:
Good to know.