intel/node-realsense #174: 0.10.0 Release task

DonnaWuDongxia, 2017-03-27:
- [x] Cleanup 0.10.0 [issues](https://github.com/01org/node-realsense/milestone/2) except this one.
- [x] Green light from QA on quality and PnP - assignee: @haoyunfeix
- [x] Documentation - common - assignee: Kenny & Lisha
- [x] npmjs.com publish (use personal account) - assignee: Kenny
- [x] tag, release notes, source code with test cases tar ball
- [x] quick test to make sure `npm install` tar ball works
- [x] quick test to make sure installed node add-on works
- [x] sample source code tar ball
- [x] drop to internal server (powerbuilder)
- [x] notify stakeholders

kenny-y, 2017-03-27:
BTW, developer team confirms that there will be no feature merge this week (code freeze ~ release), so we won't bother creating a standalone release branch at this time.

DonnaWuDongxia, 2017-03-30:
Checked with @haoyunfeix , we need one more day to verify multi model usage case.