intel/node-realsense #180: Buildbot fail to make

halton, 2017-03-27:
https://travis-ci.org/01org/node-realsense/builds/215433077

kenny-y, 2017-03-28:
https://travis-ci.org/01org/node-realsense/builds/215436218#L2118

```
  CXX(target) Release/obj.target/realsense_object_recognition/common/camera-delegate/camera_delegate.o

In file included from ../common/camera-delegate/camera_delegate.cpp:5:0:

../common/camera-delegate/camera_delegate.h:8:31: fatal error: librealsense/rs.hpp: No such file or directory

compilation terminated.
```

`librealsense` headers should be already installed... what is going on, I wonder...

tingshao, 2017-04-05:
In fact, it was not installed.
I went through and compared some of the failed and succeeded builds.
failed builds like this one, is due to:
`Err:115 http://realsense-alm-public.s3.amazonaws.com/apt-repo xenial/main amd64 librealsense-slam2 amd64 2.0.4-0ubuntu1~25.gbp73dfd6
  Hash Sum mismatch`
this failure to download the package cause most of the realsense related packages not installed.

A previous failure of build No 33, is caused by hash mismatch when downloading librealsense-samples package:
`Err:116 http://realsense-alm-public.s3.amazonaws.com/apt-repo xenial/main amd64 librealsense-samples amd64 0.6.5-1ubuntu1~21.gbp8e624e
  Hash Sum mismatch`

halton, 2017-04-05:
Checked with @tingshao for recent buildbot logs at https://travis-ci.org/01org/node-realsense/builds, this failure is occasionally happening.  So close it as not fix.

closed 2017-04-05 by halton