intel/node-realsense #181: [PT][OR] Core dumped when invoke start() method on ZR300

haoyunfeix, 2017-03-27:
For example:
`node example/pt-internal.js`
Result:
```
Logger library successfully loaded: librealsense_logger.so
log4cxx: Could not read configuration file [/home/yunfei/realsense/logs//rslog.properties].
log4cxx: No appender could be found for logger (root).
log4cxx: Please initialize the log4cxx system properly.
------------------ state:  ready
------------ Start
------------------ state:  ready
------------------ state:  ready
------------------ state:  ready
------------------ state:  ready
------------  start  done
...
Segmentation fault (core dumped)
```
Result Matrix:

Camera | Release | Result
---|---|---
ZR300 | 0.9.1 | :white_check_mark:
R200 | 0.9.1 | :white_check_mark:
ZR300 | 0.10.0 | :x:
R200 | 0.10.0 | :white_check_mark:

haoyunfeix, 2017-03-27:
This issue doesn't reproduce when install from tar ball.

tingshao, 2017-03-28:
@haoyunfeix Do you mean it crashes only when run the pt-internal.js inside the person-tracking source tree?

haoyunfeix, 2017-03-28:
@tingshao, all test cases that `require('bindings')('realsense_person_tracking')` will crash, pt-internal.js is an example.
But if I use `tar -C /path/to/node-realsense/src/ -hzcvf pt.tar.gz person-tracking && npm install ./pt.tar.gz` and change test cases to `require('node-person')`, it works.

tingshao, 2017-03-28:
Thanks, I will try a zr300 then.

kenny-y, 2017-03-28:
It looks like fisheye stream enabling is the root cause, `32720x32720:RAW8@0Hz FISHEYE`

Related to #163

tingshao, 2017-03-28:
The crash using pt-internal.js seems a separate crash issue. And only happened using the old zr300 and related with motion event.

tingshao, 2017-03-28:
It's the callstack.

#0  0x00007ffff1a6a334 in std::_Function_base::_M_empty (this=0xfb17e8000010c0) at /usr/include/c++/5/functional:1834
#1  0x00007ffff1a6c659 in std::function<void (rs::motion_data)>::operator()(rs::motion_data) const (this=0xfb17e8000010c0, __args#0=...) at /usr/include/c++/5/functional:2265
#2  0x00007ffff1a69127 in CameraDelegateDevice::<lambda(rs::motion_data)>::operator()(rs::motion_data) const (__closure=0x7fffd4018ac8, data=...)
    at ../common/camera-delegate/camera_delegate.cpp:238
#3  0x00007ffff1a69b6a in std::_Function_handler<void(rs::motion_data), CameraDelegateDevice::enable_motion_tracking(std::function<void(rs::motion_data)>, std::function<void(rs::timestamp_data)>)::<lambda(rs::motion_data)> >::_M_invoke(const std::_Any_data &, <unknown type in /home/realsense/node-realsense/src/person-tracking/build/Debug/realsense_person_tracking.node, CU 0x2cd9d, DIE 0x4211f>) (__functor=..., 
    __args#0=<unknown type in /home/realsense/node-realsense/src/person-tracking/build/Debug/realsense_person_tracking.node, CU 0x2cd9d, DIE 0x4211f>)
    at /usr/include/c++/5/functional:1871
#4  0x00007ffff1a6c682 in std::function<void (rs::motion_data)>::operator()(rs::motion_data) const (this=0x7fffd4018ac8, __args#0=...) at /usr/include/c++/5/functional:2267
#5  0x00007ffff1a6a9eb in rs::motion_callback::on_event (this=0x7fffd4018ac0, e=...) at /usr/include/librealsense/rs.hpp:371
#6  0x00007ffff16deabc in ?? () from /usr/lib/x86_64-linux-gnu/librealsense.so.1
#7  0x00007ffff174a29c in rsimpl::uvc::subdevice::poll_interrupts(libusb_device_handle*, std::vector<rsimpl::uvc::subdevice*, std::allocator<rsimpl::uvc::subdevice*> > const&, unsigned short) () from /usr/lib/x86_64-linux-gnu/librealsense.so.1
#8  0x00007ffff174a491 in std::thread::_Impl<std::_Bind_simple<rsimpl::uvc::device::start_data_acquisition()::{lambda()#1} ()> >::_M_run() ()
   from /usr/lib/x86_64-linux-gnu/librealsense.so.1
#9  0x00007ffff6204c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007ffff757c6fa in start_thread (arg=0x7fffdbfff700) at pthread_create.c:333
#11 0x00007ffff596ab5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

tingshao, 2017-03-28:
I found, When pt runs, motion event is switched on, and motion event call backs can be received. when pt resets (stop first), these call backs are stopped too. but when PT start again, it crashed in a motion event callback.
I guess the way we manipulate motion event source is in a wrong way. Is there something missing in the following steps:
1. device->enable_motion_tracking(callback, callback2)
2. device->start(rs::all_sources)
3. device->stop(rs::all_sources);
4. device->enable_motion_tracking(callback, callback2)
5. device->start(rs::all_sources); // crash afther this.

if we don't enable rs::all_source, and only enable rs::video which the current PT addon is, the program would crash after stop() is called at step 3.

kenny-y, 2017-03-29:
Preliminary finding -- maybe a call to `disable_motion_tracking()` is required before calling `stop()`

tingshao, 2017-03-29:
I tried this change, the three addon's mocha works on PC, but some or cases failed on Joule.

PR #186 (2017-03-30): Fix zr300 core dump
Closes #181
closed 2017-03-30 by tingshao