intel/node-realsense #182: [LRS] Throw TypeError when invoking method with negative arguments

haoyunfeix, 2017-03-28:
Pattern 1: When number of arguments is not correct
e.g. 
```
  it('Negative - Call getName with 1 option', function() {
    return new Promise((resolve, reject) => {
      device.getName('dummy').then((name) => {
        reject('Should get reject but got resolved promise');
      }).catch((e) => {
        resolve();
      });
    });
  });
```
Result:
```
  1) Test Device interfaces Negative - Call getName with 1 option:
     TypeError: Cannot read property 'then' of undefined
      at test-getName.js:23:30
      at Context.<anonymous> (test-getName.js:22:12)
```
Pattern 2: When value of argument is not correct
e.g.
```
  it('Negative - Call supportsOption with illegal options', function() {
    return new Promise((resolve, reject) => {
      device.supportsOption('dummy').then((enable) => {
        reject('Should get reject but got resolved promise');
      }).catch((e) => {
        resolve();
      });
    });
  });
```
Result:
```
  3) Test Device interfaces Negative - Call supportsOption with illegal options:
     TypeError: Illegal argument: OptionType option == 'dummy'
      at TypeError (native)
      at test-supportsOption.js:49:14
      at Context.<anonymous> (test-supportsOption.js:48:12)
```

Expected Result: Should get reject status promise when met error.

minggangw, 2017-03-28:
I'll take this

minggangw, 2017-03-28:
The pattern 1 issue is caused by the widl-nan tool, which generates the binding codes. This not only happens in librealsense, but also other modules I think. A [issue](https://github.com/01org/widl-nan/issues/96) has been created, so I think we'd better fix this by changing the logics of the tool.

DonnaWuDongxia, 2017-03-30:
Move out of this milestone