intel/node-realsense #190: [Multi-models] Core dump when invoke SLAM method when running PT or OR modules.

haoyunfeix, 2017-03-30:
Modules|Result
---|---
PT + OR | Pass
PT + SLAM | Fail
PT + SLAM | Fail
PT + OR + SLAM | Fail

Environment:
Joule 570x Ubuntu 16.04 + Realsense ZR300
PC Ubuntu 16.04 + Realsense ZR300

Test cases:
https://github.com/haoyunfeix/node-realsense-1/blob/2cdaf0d5b790d3a5b961479fa1cf2572db35b40a/src/test/test-or-slam.js

Result:
`Segmentation fault (core dumped)`

kenny-y, 2017-04-01:
Some findings:

The first test case:
```
  it('Make sure SLAM + OR can be run/stop at (almost) the same time', function() {
    // eslint-disable-next-line
    this.timeout(60 * 1000);
    return Promise.all([verifySLAM(100), verifyOR(100)]);
  });
```
will timeout due to no `tracking` event of SLAM module will be triggered, and therefore the rest of the cases won't be able to run successfully (because SLAM module is still running after the timeout), the following is the output of my console (log lines added to the test suite)

```
run OR
SLAM created
run SLAM
SLAM started
stopping OR
OR is stopped
```

Now I'm tracking down the reason why `tracking` event of SLAM module is not sent.

kenny-y, 2017-04-01:
Follow-up finding:
`fisheye` should be enabled to run SLAM module.

```
const cameraOptions = {
  color: {
    width: 640,
    height: 480,
    frameRate: 30,
    isEnabled: true,
  },
  fisheye: {
    width: 640,
    height: 480,
    frameRate: 30,
    isEnabled: true,
  },
  depth: {
    width: 320,
    height: 240,
    frameRate: 30,
    isEnabled: true,
  },
};

function runSLAM(slam) {
  console.log('run SLAM');
  return slam.setCameraOptions(cameraOptions).then(function() {
    return slam.start();
  });
}

function runOR(or) {
  return or.setObjectRecognitionOptions({
    mode: 'single-recognition',
    confidenceThreshold: 0.3,
    computeEngine: 'CPU',
    enableSegmentation: true,
    maxReturnObjectCount: 2,
    enableObjectCenterEstimation: false,
  }).then(function() {
    return or.setCameraOptions(cameraOptions);
  }).then(function() {
    console.log('run OR');
    return or.start();
  });
}
```
After enabling this, the first test case can pass, investigating the rest.

kenny-y, 2017-04-01:
in the 2nd test case, the call to camera control interface `enable_motion_tracking()` will hang and never return, causing test cases to timeout...

kenny-y, 2017-04-01:
`enable_motion_tracking()` is not the only that will hang, other "plain" functions can also hang, e.g. `get_device_count()`. I guess camera is out of control due to some reason... need to find the root cause.

kenny-y, 2017-04-01:
Suggest to close this issue -- the root cause of this issue is that the `CameraOptions` is not set to "fisheye enabled"

And then open up a new issue -- describe the never return librealsense API call after the below sequence
1. Start [OR|PT] + SLAM
2. Stop [OR|PT] + SLAM
3. Start [OR|PT] + SLAM
4. In one step of starting camera, an API call will hang.

kenny-y, 2017-04-01:
Opening #194 and closing this one.

closed 2017-04-01 by kenny-y