intel/node-realsense #191: [Sample] pt_tutorial_4_web does not recognize the person which in database

haoyunfeix, 2017-03-31:
Steps:
1. `node main.js` and `open browser *.*.*.*:8000/view.html`
2. Mouse click tracked person on display for registering and write into database file
3. Clicking on the load database button. 

Expects result:
Loads the database that has information about the recognized persons and displays the rid of already registered person when they are present in the FOV.

Actually result:
Seems like it didn't recognize person and didn't show the rid of registered person.

tingshao, 2017-04-05:
@haoyunfeix This demo is a bit difficult to use, Though I was a bit confused with its implementation, I have to make it consistent with its native version.  here I have some tips you may check before run it.
1. Person is registered in database during the the 'persontracked' event callback, though it was called repeatedly, the duplicated operation would take no effect.
2. once the person is registered, its rid would appear.
3. the database is stored into file only when you click one person.

I made a test and you can try it too:
1. run main.js, and make sure my head is registered and the rid appears (no operation needed except make sure you face toward the camera in a good distance)
2. click my head to store the database.
3. stop the program. then you can find a `person_recognition_id.db` file in current directory. We need it in the next program.
4. run another test program test.js, (which doesn't register person, but only recognize person using old database.) click load database button, then move around your head, you can see you will be recognized.
the test.js is in attachement named main.txt.
[main.txt](https://github.com/01org/node-realsense/files/895056/main.txt)

tingshao, 2017-04-17:
@haoyunfeix How is it going with the steps described in the last reply? Some details may be helpful. thanks.

haoyunfeix, 2017-04-17:
This works for me, thanks.

> This demo is a bit difficult to use, Though I was a bit confused with its implementation

Yes, I also tried the native sample. That confused me too because it forced to register people immediately when people tracked. Mouse click seems only to export the database as a file not to register people.

I tried to update the js code to implement the description as below at https://github.com/thewebera/realsense-ocf-demos/tree/master/demo2

> Mouse click in any of the detected person's rectangle, activates registration for that person and saves the information in a database (database name currently hardcoded) with a Recognition ID (rid) assigned to that person. 

Anyway, we can close this issue now.

closed 2017-04-17 by haoyunfeix