intel/node-realsense #221: [Multi-models] SLAM stopped working after PT is stopped.

haoyunfeix, 2017-04-06:
Steps:
1. Enabled fisheye for PT and SLAM module and create instance.
2. Run PT and SLAM at the same time.
3. Stop PT module while SLAM is running.

Result:
`Segmentation fault (core dumped)`

tingshao, 2017-04-17:
@haoyunfeix Is this issue still valid after the configuration tips we mentioned in last sync meeting?

haoyunfeix, 2017-04-17:
@tingshao, yes, still valid.
Please help to check if I do something wrong at https://github.com/haoyunfeix/node-realsense-1/blob/1aca21bcbf883e297a48bcbc7f9bc008f62b58f3/src/test/test-pt-slam.js

haoyunfeix, 2017-04-18:
Also you can simply to reproduce this issue with code https://github.com/01org/node-realsense/blob/master/src/example/run-or-pt-slam.js
Step 1: change `runPT()` to `runPT(5000)`
Step 2: `node run-or-pt-slam.js`

tingshao, 2017-04-18:
-------------------------------------------------
Timeout, stopping person-tracking module.
-------------------------------------------------
SLAM: ontracking:  EventInfo {
  cameraPose: 
   [ 0.43756812810897827,
     -0.8991833925247192,
     0.0018310844898223877,
     0.22521911561489105,
     0.898486852645874,
     0.4371471703052521,
     -0.04029655456542969,
     2.6658077239990234,
     0.035433560609817505,
     0.01927769184112549,
     0.9991861581802368,
     0.3839190602302551,
     0,
     0,
     0,
     1 ],
  accuracy: 'low' }
OR: frameprocessed
OR: recognition [ RecognitionInfo { probability: 0.9973440766334534, label: 'sofa' } ]
SLAM: ontracking:  EventInfo {
  cameraPose: 
   [ 0.4376210570335388,
     -0.8991577625274658,
     0.001786559820175171,
     0.22572417557239532,
     0.8984590768814087,
     0.4372009038925171,
     -0.040332674980163574,
     2.6525211334228516,
     0.03548434376716614,
     0.01925557851791382,
     0.9991847276687622,
     0.3837142884731293,
     0,
     0,
     0,
     1 ],
  accuracy: 'low' }
OR: frameprocessed
OR: recognition [ RecognitionInfo { probability: 0.9978131055831909, label: 'sofa' } ]
SLAM: onerror:  error: Junk time stamp in stream:4	with frame counter:161
SLAM: onerror:  error: Junk time stamp in stream:0	with frame counter:155
OR: frameprocessed
OR: recognition [ RecognitionInfo { probability: 0.9968453049659729, label: 'sofa' } ]
SLAM: onerror:  error: Junk time stamp in stream:0	with frame counter:156

Thread 15 "node" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc7148700 (LWP 31177)]
0x00007fffbe4c9f80 in ?? ()
(gdb) bt
#0  0x00007fffbe4c9f80 in ?? ()
#1  0x00007fffe32f0b18 in std::_Function_handler<void (rs::motion_data), CameraDelegateDevice::enable_motion_tracking(std::function<void (rs::motion_data)>, std::function<void (rs::timestamp_data)>)::{lambda(rs::motion_data)#1}>::_M_invoke(std::_Any_data const&, rs::motion_data&&) () from /home/tshao/work/code/01org/node-realsense/src/person-tracking/build/Release/realsense_person_tracking.node
#2  0x00007fffe32f09d6 in rs::motion_callback::on_event(rs_motion_data) () from /home/tshao/work/code/01org/node-realsense/src/person-tracking/build/Release/realsense_person_tracking.node
#3  0x00007fffefa0fabc in ?? () from /usr/lib/x86_64-linux-gnu/librealsense.so.1
#4  0x00007fffefa7b29c in rsimpl::uvc::subdevice::poll_interrupts(libusb_device_handle*, std::vector<rsimpl::uvc::subdevice*, std::allocator<rsimpl::uvc::subdevice*> > const&, unsigned short) ()
   from /usr/lib/x86_64-linux-gnu/librealsense.so.1
#5  0x00007fffefa7b491 in std::thread::_Impl<std::_Bind_simple<rsimpl::uvc::device::start_data_acquisition()::{lambda()#1} ()> >::_M_run() () from /usr/lib/x86_64-linux-gnu/librealsense.so.1
#6  0x00007ffff6204c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff757c6ba in start_thread (arg=0x7fffc7148700) at pthread_create.c:333
#8  0x00007ffff596a82d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb)