intel/node-realsense #225: Packages failed to install via "npm install node-*"

haoyunfeix, 2017-04-18:
Environment: 

Platform | PC | Joule
--- | --- | ---
node| 6.9.1| 4.2.6
npm| 3.10.8| 3.5.2

Run: `npm install node-person` `npm install node-object` `npm install node-slam`

Then failed in compilation:

```
yunfei@yunfei-sdf:~/tmp$ npm install node-person
npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated to-iso-string@0.0.2: to-iso-string has been deprecated, use @segment/to-iso-string instead.
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue

> node-person@0.10.0 install /home/yunfei/node_modules/node-person
> node -e "require('./common/scripts/build_utils.js').runNodeGyp();"

(node:885) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): TypeError: Cannot read property 'on' of null
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/yunfei/node_modules/node-person/build'
  COPY Release/obj.target/realsense_person_tracking/geni/process_idl.js
  TOUCH binding_gyp_realsense_person_tracking_target_widl_gen.intermediate
  ACTION Generating NAN C++ code from WIDL files binding_gyp_realsense_person_tracking_target_widl_gen.intermediate
WebIDLParseError {
  message: 'No end of extended attribute',
  line: 4,
  input: '\n]\ninterface ',
  tokens:
   [ { type: 'whitespace', value: '\n' },
     { type: 'other', value: ']' },
     { type: 'whitespace', value: '\n' },
     { type: 'identifier', value: 'interface' },
     { type: 'whitespace', value: ' ' } ] }
  CXX(target) Release/obj.target/realsense_person_tracking/addon.o
  CXX(target) Release/obj.target/realsense_person_tracking/bounding_box2d_info.o
  CXX(target) Release/obj.target/realsense_person_tracking/common/camera-delegate/camera_delegate.o
  CXX(target) Release/obj.target/realsense_person_tracking/common/camera-delegate/camera_delegate_instance.o
  CXX(target) Release/obj.target/realsense_person_tracking/common/camera-options/camera_options_host.o
  CXX(target) Release/obj.target/realsense_person_tracking/common/camera-options/camera_options_host_instance.o
  CXX(target) Release/obj.target/realsense_person_tracking/common/extrinsics.o
  CXX(target) Release/obj.target/realsense_person_tracking/common/frame_data_assembler.o
../addon.cpp:5:42: fatal error: gen/nan__bounding_box2d_info.h: No such file or directory
#include "gen/nan__bounding_box2d_info.h"
                                          ^
compilation terminated.
In file included from ../bounding_box2d_info.h:13:0,
                 from ../bounding_box2d_info.cpp:5:
../common/geometry/rect2d.h:14:34: fatal error: gen/generator_helper.h: No such file or directory
#include "gen/generator_helper.h"
                                  ^
compilation terminated.
realsense_person_tracking.target.mk:235: recipe for target 'Release/obj.target/realsense_person_tracking/addon.o' failed
make: *** [Release/obj.target/realsense_person_tracking/addon.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from ../common/extrinsics.cpp:5:0:
.././common/extrinsics.h:14:34: fatal error: gen/generator_helper.h: No such file or directory
#include "gen/generator_helper.h"
                                  ^
compilation terminated.
In file included from ../common/camera-options/camera_options_io.h:12:0,
                 from ../common/camera-options/camera_options_host.h:12,
                 from ../common/camera-options/camera_options_host.cpp:7:
../common/camera-options/camera_options_type.h:8:32: fatal error: gen/camera_options.h: No such file or directory
#include "gen/camera_options.h"
                                ^
compilation terminated.
In file included from ../common/frame_data_assembler.h:15:0,
                 from ../common/frame_data_assembler.cpp:5:
../common/frame_data.h:13:34: fatal error: gen/generator_helper.h: No such file or directory
#include "gen/generator_helper.h"
                                  ^
compilation terminated.
In file included from ../common/camera-options/camera_options_io.h:12:0,
                 from ../common/camera-options/camera_options_host.h:12,
                 from ../common/camera-options/camera_options_host_instance.h:8,
                 from ../common/camera-options/camera_options_host_instance.cpp:5:
../common/camera-options/camera_options_type.h:8:32: fatal error: gen/camera_options.h: No such file or directory
#include "gen/camera_options.h"
                                ^
compilation terminated.
realsense_person_tracking.target.mk:235: recipe for target 'Release/obj.target/realsense_person_tracking/bounding_box2d_info.o' failed
make: *** [Release/obj.target/realsense_person_tracking/bounding_box2d_info.o] Error 1
realsense_person_tracking.target.mk:235: recipe for target 'Release/obj.target/realsense_person_tracking/common/extrinsics.o' failed
make: *** [Release/obj.target/realsense_person_tracking/common/extrinsics.o] Error 1
realsense_person_tracking.target.mk:235: recipe for target 'Release/obj.target/realsense_person_tracking/common/frame_data_assembler.o' failed
make: *** [Release/obj.target/realsense_person_tracking/common/frame_data_assembler.o] Error 1
realsense_person_tracking.target.mk:235: recipe for target 'Release/obj.target/realsense_person_tracking/common/camera-options/camera_options_host.o' failed
make: *** [Release/obj.target/realsense_person_tracking/common/camera-options/camera_options_host.o] Error 1
realsense_person_tracking.target.mk:235: recipe for target 'Release/obj.target/realsense_person_tracking/common/camera-options/camera_options_host_instance.o' failed
make: *** [Release/obj.target/realsense_person_tracking/common/camera-options/camera_options_host_instance.o] Error 1
rm binding_gyp_realsense_person_tracking_target_widl_gen.intermediate
make: Leaving directory '/home/yunfei/node_modules/node-person/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.0-040400-generic
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild" "-j" "8"
gyp ERR! cwd /home/yunfei/node_modules/node-person
gyp ERR! node -v v6.9.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/home/yunfei/package.json'
npm WARN yunfei No description
npm WARN yunfei No repository field.
npm WARN yunfei No README data
npm WARN yunfei No license field.
npm ERR! Linux 4.4.0-040400-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "node-person"
npm ERR! node v6.9.1
npm ERR! npm  v3.10.8
npm ERR! code ELIFECYCLE

npm ERR! node-person@0.10.0 install: `node -e "require('./common/scripts/build_utils.js').runNodeGyp();"`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-person@0.10.0 install script 'node -e "require('./common/scripts/build_utils.js').runNodeGyp();"'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-person package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node -e "require('./common/scripts/build_utils.js').runNodeGyp();"
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-person
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-person
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/yunfei/tmp/npm-debug.log
```

tingshao, 2017-04-18:
That's because the webidl2 packaged depended by widl-nan released 2.4.0 recently, and [a white-space related bug](https://github.com/w3c/webidl2.js/issues/64) was triggered. We may fall back to 2.3.0 in widl-nan to avoid this failure, and revert back to latest version after it fixed.

tingshao, 2017-04-18:
Now we have two choices:
1. Remove some '\n' characters in geometry.widl:
change below code
```
[
constructor
..
]
```
to
```
[constructor
...]
```
2. use 2.3.0 of webidl2
I prefer choice 1, then we won't need to depend on this webidl2 bug, and could also make our code clean.

tingshao, 2017-04-18:
@halton @kenny-y As we have two choices, to release a new version of widl-nan or our node-xxx. Which do you prefer?

closed 2017-04-25 by kenny-y

haoyunfeix, 2017-04-25:
Confirmed, works with `npm install node-person[object|slam]`.

commit 7d2268b1 (2017-04-25): Closes #225

Change IDL file to adapt the change of dependent modules